ArmDeveloperEcosystem/lorawan-library-for-pico #3: LoRaWAN, Raspberry Pi Pico, HOPERF Module RFM95W, The Things Network V3

jbinko, 2021-06-16:
I wanted to document the configuration steps for European region, TTNv3 and for RFM95.
Based on the articles here:
https://www.raspberrypi.org/blog/how-to-add-lorawan-to-raspberry-pi-pico/
https://www.hackster.io/sandeep-mistry/connect-your-adafruit-feather-rp2040-to-the-things-network-5c0c84

I started with draft of the documentation here:
https://github.com/jbinko/jbinko.github.io/tree/master/RFM95W

But I cannot make it run. 
I did exactly what I wrote in the documentation and I believe it is correct based on articles above.
Not sure if I’m missing something specific for European region (like MAC version, Regional Parameters, Tx Power, etc), or source code of pico lorawan project changed meanwhile, TTN changed meanwhile and is not compatible. Not sure.
I tried couple of options/tweaks but it is not able to join TTN network. It always timeouts on Joining LoRaWAN network.
If I use the same RFM module with Arduino + Arduino library I can connect with TTN just fine.
So HW should be OK.

I also received different value for JoinEUI but I guess it is OK and how the library works.
Maybe some more detailed debugging option can help.

Please, if you have any guidance what might be wrong, please let me know.
I guess it is some detail which I didn’t find for my last two weeks of explorations.

Pico LoRaWAN - OTAA - Temperature + LED

Initilizating LoRaWAN ... DevEui      : E6-60-58-38-XX-XX-XX-XX
JoinEui     : 3B-3A-01-10-XX-XX-XX-XX
Pin         : 00-00-00-00

success!
Joining LoRaWAN network ...
###### =========== MLME-Request ============ ######
######               MLME_JOIN               ######
###### ===================================== ######
STATUS      : OK

###### =========== MLME-Confirm ============ ######
STATUS      : Rx 2 timeout

sandeepmistry, 2021-06-25:
Hi @jbinko,

Have you looked at the "Live Data" tab of the The Things Stack console for any tips, here's what I see:

[image]

Let me know what you find there.

jbinko, 2021-06-30:
Hey Sandeep,
Thank you for response.

Yeah the challenge is it never connects with TTN successfully for whatever reason, so I do not see any communication data in TTN network.
I’m suspecting EU specific configuration might not work or TTNv3 might be incompatible.
Any ideas what I can try to troubleshoot? Any advanced tracing?

Thanks

sandeepmistry, 2021-07-05:
Hi @jbinko,

Do you have access to the gateway in the The Things Stack console to see if it is receiving the packets?

Can you confirm that you are not seeing any "Join-request" entries in the screenshot I mentioned in https://github.com/sandeepmistry/pico-lorawan/issues/3#issuecomment-868752704 ?

closed 2021-10-05 by sandeepmistry

Nivaldex, 2022-01-05:
Hello guys I have a problem that I believe to be the subject of this conversation.

a) I was able to get a gateway (actually a repeater) based on ESP-32 (https://github.com/things4u/ESP-1ch-Gateway) to communicate with TTN v3;
b) It sends data from a node made with the HopeHF and arduino mini;
c) But I can't receive these data in the application with a end device. I hope to be able to send the data to AllThingsTalk Maker.

Does anyone have any idea where I'm going wrong. The system worked perfectly in version 2, however, in version 2 I didn't need to use the gateway.

Thanks.

[image]

AiyionPrime, 2022-01-15:
@sandeepmistry I just got started using @jbinko's instructions on how to configure the pico, the source and how to wire up the lora breakout board: https://github.com/jbinko/jbinko.github.io/tree/master/RFM95W

I can reproduce the issue described above, I'm hitting the same rx2 timeout.
A friend of mine has a gateway, ad as i visited him yesterday, we inspected his gatewaylog as the temperature or hello otaa example ran.

Most of the joins are not recognized at all, the one or two that are, show of with a critically bad SNR.
I think the issues @jbinko found first, has not been resolved yet and we should reopen this issue.

I'm interested in providing more testresults if wanted.

sandeepmistry, 2022-01-18:
@AiyionPrime can you please try the changes from PR #16 ?

AiyionPrime, 2022-01-18:
I did. It did not work, though.

```Pico LoRaWAN - OTAA - Temperature + LED

Initilizating LoRaWAN ... 
###### =========== CTXS RESTORED =========== ######
Size        : 2208

DevEui      : 70-B3-D5-7E-D0-04-B3-9B
JoinEui     : 00-00-00-00-00-00-00-00
Pin         : 00-00-00-00

success!
Joining LoRaWAN network ...
###### =========== MLME-Request ============ ######
######               MLME_JOIN               ######
###### ===================================== ######
STATUS      : OK
.......
###### =========== MLME-Confirm ============ ######
STATUS      : Rx 2 timeout

###### =========== MLME-Request ============ ######
######               MLME_JOIN               ######
###### ===================================== ######
STATUS      : OK
........
###### =========== MLME-Confirm ============ ######
STATUS      : Rx 2 timeout

###### =========== MLME-Request ============ ######
######               MLME_JOIN               ######
###### ===================================== ######
STATUS      : OK
........
###### =========== MLME-Confirm ============ ######
STATUS      : Rx 2 timeout

###### =========== MLME-Request ============ ######
######               MLME_JOIN               ######
###### ===================================== ######
STATUS      : OK
.......
###### =========== MLME-Confirm ============ ######
STATUS      : Rx 2 timeout

###### =========== MLME-Request ============ ######
######               MLME_JOIN               ######
###### ===================================== ######
STATUS      : OK
..
```

gniezen, 2022-01-21:
I also struggled with the "Rx 2 timeout" issue. Either moving my node closer to the gateway or restarting The Things Indoor Gateway (TTIG) fixed it somehow :man_shrugging:

AiyionPrime, 2022-01-22:
@gniezen We sat about 10 meters away from the gateway and barely got anything through.
we swapped the lora module for one we knew he had deployed successfully earlier.

As I bought the pi pico was bought for this project, there's not much besides this stack or a misconfiguration, that could be wrong on my end.
I'll paste my changes to the code base later on for you to review.

AiyionPrime, 2022-01-22:
@gniezen You tried to use this in UK, I assume? I'm trying to get it working in Hanover, DE.

sandeepmistry, 2022-02-01:
@AiyionPrime there's a conversion in https://github.com/ArmDeveloperEcosystem/lorawan-library-for-pico/issues/14#issuecomment-1017049643 about needing to erase the NVM. There's new example in PR #18 you can try to do this.

AiyionPrime, 2022-02-02:
@sandeepmistry I just did, flashed the example and ran it (likely) successfully, as the tty reported.

Afterwords I compiled and ran hello_otaa again, which did not change its behavior: https://bpa.st/AOXQ

The restriction which would eventually follow is different to #14, as it does not come up at the start, but after I throw a lot of join messages on the band.

tnakasaki, 2022-02-02:
@AiyionPrime I'm not sure about your signal strength issues. But if you erase Lorawan context from memory, DevNonce will reset. LoRaMac-node will then use DevNonce values starting from 0, and your Lorawan network may not accept join attempts made with an already used DevEui/JoinEui/DevNonce combination. Have you tried generating a new device with a new DevEui and sending join requests to that?

placekowo, 2023-06-23:
I've had this issue, both using a Pico + Adafruit RFM95 breakout, and an Adafruit Feather RP2040 RFM95. Clearing the NVM didn't help.
I've found that setting the PACONFIG register to always be ```0xFF``` got the module to connect to my gateway reliably. 
I'll investigate this more, but it seems like for me the issue was something related to transmit power. I got it to connect ocassionally before, but with a very bad SNR, even with the node right next to the gateway.

In ```sx1276-board.c```
```c
void SX1276SetRfTxPower( int8_t power )
{
    ...
    //SX1276Write( REG_PACONFIG, paConfig );
    SX1276Write( REG_PACONFIG, 0xFF );
    SX1276Write( REG_PADAC, paDac );
}
```

sandeepmistry, 2023-06-30:
@placekowo thanks for the comment, maybe `SX1276GetPaSelect` needs to be changed to return a fixed value?

https://github.com/ArmDeveloperEcosystem/lorawan-library-for-pico/blob/f18fec783f13575d9cf9a27cef65a150b12c0fc0/src/boards/rp2040/sx1276-board.c#L205-L215

zackfuchtel, 2023-09-25:
Exactly my problem. I am able to look on my own HeliumGW. Not a single bit is sended by the PI / RFM95W combo. Nothing to see in the logs on the Gateway. Another piece with d1mini & RFM95W ist working fine. What the...?

kripton, 2024-07-23:
I experienced the same issue on a iLabs Challenger RP2040 LoRa (https://ilabs.se/challenger-rp2040-lora-datasheet). Hardcoding the register value to 0xFF fixes the problem.

CR1N993R, 2024-07-23:
Setting it directly to 0xFF also fixed the issue for me however in the debug log it still says that tx power is 0. So there is probably a proper way to solve this.

###### =====   UPLINK FRAME        3   ===== ######

CLASS       : A

TX PORT     : 2
TX DATA     : UNCONFIRMED
20 

DATA RATE   : DR_4
U/L FREQ    : 868100000
TX POWER    : 0
CHANNEL MASK: 00FF

morris-murigi, 2024-09-01:
Hello did anyone solve this issue?
11:19:08.584 -> Pico LoRaWAN - OTAA - Temperature + LED
11:19:08.584 -> 
11:19:08.584 -> Initilizating LoRaWAN ... 
11:19:08.584 -> ###### =========== CTXS RESTORED =========== ######
11:19:08.584 -> Size        : 2208
11:19:08.584 -> 
11:19:08.631 -> DevEui      : 70-B3-D5-7E-D0-06-9E-FB
11:19:08.631 -> JoinEui     : 12-34-56-78-91-23-45-67
11:19:08.631 -> Pin         : 00-00-00-00
11:19:08.631 -> 
11:19:08.631 -> success!
11:19:08.631 -> Joining LoRaWAN network ...
11:19:08.631 -> ###### =========== MLME-Request ============ ######
11:19:08.631 -> ######               MLME_JOIN               ######
11:19:08.631 -> ###### ===================================== ######
11:19:08.631 -> STATUS      : OK
11:19:09.620 -> .........
11:19:18.151 -> ###### =========== MLME-Confirm ============ ######
11:19:18.151 -> STATUS      : Rx 1 timeout
11:19:18.151 -> 
11:19:18.151 -> ###### =========== MLME-Request ============ ######
11:19:18.151 -> ######               MLME_JOIN               ######
11:19:18.151 -> ###### ===================================== ######
11:19:18.151 -> STATUS      : OK

jp-pressac, 2025-06-13:
> I experienced the same issue on a iLabs Challenger RP2040 LoRa (https://ilabs.se/challenger-rp2040-lora-datasheet). Hardcoding the register value to 0xFF fixes the problem.

@kripton 

I am jsut picking up the challenger and have tried this not getting any messages through to my gateway. Using Chirp stack and not TTN though, just wondering if you have any suggestions?

morris-murigi, 2025-06-13:
When I faced the challenge, I was using the Pico C and C++ SDK, as well as the lorawan-library-for-pico. However, I switched to the Arduino Framework, using the Pico board definition for Arduino, and the Arduino-LMIC library ("MCCI LoRaWAN LMIC Library"), and it worked flawlessly.