ArmDeveloperEcosystem/lorawan-library-for-pico #8: Sending wrong Device EUI and Join EUI

simeonyetarian, 2021-07-28:
I've entered my Device EUI, Join EUI and App Key in config.h of the otaa_temperatur_led example but the data that the gateway receives contains a different Device EUI and Join EUI (which are both the same for some reason).

My config.h file:
[image]

Minicom output of the Raspberry Pico with RFM95W:
(This already shows the wrong Device EUI and Join EUI)
[image]

Live data from the gateway with the wrong Device EUI and Join EUI:
[image]

I am using The Things Network V3 with EU868 configuration.
I've also tried to change LoraWAN MAC version and Regional parameters version in the TTN console, but this has no impact since the minicom output of the pico already shows the wrong Device EUI and Join EUI.

Is there anything I can do to fix this?

tnakasaki, 2021-07-28:
I have been having the same issues, using US915 settings. I will post screenshots when I have time.

This issue only happens when compling on a Raspberry Pi 4. When I compile the uf2 on a Mac, the expected EUI values are transmitted.

simeonyetarian, which platform are you using to compile?

simeonyetarian, 2021-07-28:
@tnakasaki Thanks for sharing that information! Indeed, I am compiling on a Raspberry Pi 4. Maybe I will try to compile on another device then

simeonyetarian, 2021-07-28:
Update: I can confirm that compiling on a Windows machine instead of a Raspberry Pi works. 
So this seems to be specific to the Raspberry Pi

tnakasaki, 2021-07-29:
Here are some screenshots that illustrate the aforementioned issue.

config.h

[image]

**cmake** output (configuring for adafruit feather rp2040 board)

[image]

Partial **make** output

[image]

TTN gateway data. Note the incorrect values for both JoinEUI and DevEUI, 618A011029890110

[image]

Also, in config.h, if **LORAWAN_DEVICE_EUI** is set to **NULL**,  a correct DevEUI will be transmitted. That is, the same value printed in the default_dev_eui example program. JoinEUI remains incorrect.

tnakasaki, 2021-07-29:
These are screenshots of **cmake** and **make** performed on a Mac. This compilation produced an executable that functioned correctly.

**cmake**

[image]

**make**

[image]

sandeepmistry, 2021-09-20:
@simeonyetarian @tnakasaki Looks like it might be related to the version on Arm GCC used.

Based on the screenshots @tnakasaki provided in https://github.com/ArmDeveloperEcosystem/lorawan-library-for-pico/issues/8#issuecomment-889052952, they were using `7.3.1` or `8.3.0` on the Raspberry Pi, but using `9.2.1` on the Mac.

Any chance you could try to use a newer compiler version on the Raspberry Pi?

@simeonyetarian could you please provide the compiler versions you were using Raspberry Pi vs Windows? Thanks.

sandeepmistry, 2021-12-17:
Hi @simeonyetarian, @tnakasaki,

I've prepared pull request https://github.com/ArmDeveloperEcosystem/lorawan-library-for-pico/pull/12 to resolve this. It would be great if you could try out the change and provide any feedback.

ThomasLeister, 2022-01-05:
I was facing the same issue with GCC 11.2.1 on Fedora Linux (x86).
The PR seems to solve it. With debug mode enabled, the correct DevEui and JoinEui are shown.

Joining the network is not yet working, but that is probably another issue ... ;-)

sandeepmistry, 2022-01-05:
@ThomasLeister thanks for the feedback on PR #12! I've gone ahead an merged it.

closed 2022-01-05 by sandeepmistry

tnakasaki, 2022-01-06:
> Hi @simeonyetarian, @tnakasaki,
> 
> I've prepared pull request #12 to resolve this. It would be great if you could try out the change and provide any feedback.

Sorry for the late response,
I've compiled on the Raspberry Pi 4. JoinEUI and DevEUI transmit correctly. Thank you so much for your solution!

PR #12 (2022-01-05): Replace scanf %2hhx with %2x for gcc 8.3 compatibility
To resolve #8.

I've tested these changes with:

macOS
```
% arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Arm Embedded Toolchain 10-2020-q4-major) 10.2.1 20201103 (release)
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Raspberry Pi 4
```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (15:8-2019-q3-1+b1) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]
Copyright (C) 201 [... cut]