ArmDeveloperEcosystem/lorawan-library-for-pico #10: LoRa Initialisation failed

SiddhantArya, 2021-11-01:
Hi,

I've been trying to use this library for making a test node for LoRa. 
I am using RFM95W with Raspberry Pi Pico. 
I've set up the connections as mentioned in the wiki and have changed the region to IN865 in 'config.h'. Also, I am using 3.3V power out from the Pico. However, I still receive the LoRa Init failed message. 

Any help would be appreciated.

sandeepmistry, 2021-11-26:
Hi @SiddhantArya,

Are you using you trying to use this with your own LoRaWAN gateway? If so, have you looked at the gateway traffics logs in TTS web console?

I haven't used this library in India myself, but you might also need to set configure a channel mask like I had to for North America.

For example for [US902-928 ](https://www.thethingsnetwork.org/docs/lorawan/frequency-plans/#us902-928) I had to use a channel mask value of `"FF0000000000000000020000"`.

Please take a look at the TTN Frequency Plans for [IN865-867](https://www.thethingsnetwork.org/docs/lorawan/frequency-plans/#in865-867).

SiddhantArya, 2021-11-30:
Yes @sandeepmistry. I am using a pygate LoRaWAN gateway. The link of the same can be found here: [PyGate Gateway](https://pycom.io/product/pygate/). I have configured the gateway to the Indian frequency plan. The gateway shows up on the TTN console, but it doesn't show any message from my Pico + RFM95W node . I don't know where or how to set a channel mask.

wboerhout, 2021-12-20:
I am using your module in order to connect to the TTN network in the Netherlands.

Code fails in SX1276Read (does not compare to 0x12 apparently)

Anything I can do to test? Do you have any progress?

Thanks for making this possible!

sandeepmistry, 2021-12-20:
Hi @wboerhout,

Could you please open a new issue for this. Please include more details on the hardware you are using, the wiring (incl. photos), and what the  `SX1276Read` is returning.

wboerhout, 2021-12-20:
will do.

sandeepmistry, 2022-01-12:
@SiddhantArya would you be able to try ABP instead of OTAA?

sandeepmistry, 2022-04-05:
Closing for now due to lack of response.

closed 2022-04-05 by sandeepmistry