ArmDeveloperEcosystem/lorawan-library-for-pico #13: LoraWAN initialization fails in SX1276Read(REG_LR_VERSION)

wboerhout, 2021-12-20:
I have built the otaa_temperature_led example for Raspberry Pi Pico

Build goes fine, but executing, in main.c, lorawan_init_otaa fails. I added some debug code to lorawan.c, and I see that the failure occurs in the comparison `"SX1276Read(REG_LR_VERSION) != 0x12"`

As far as I can see, `SX1276Read(REG_LR_VERSION)` returns `0`

Physical setup is a Waveshare SX127x module for Raspberry Pi Pico, I don't think pictures are necessary

[lorawan.c.txt](https://github.com/ArmDeveloperEcosystem/lorawan-library-for-pico/files/7747540/lorawan.c.txt)
[main.c.txt](https://github.com/ArmDeveloperEcosystem/lorawan-library-for-pico/files/7747545/main.c.txt)
[config.h.txt](https://github.com/ArmDeveloperEcosystem/lorawan-library-for-pico/files/7747546/config.h.txt)
.

sandeepmistry, 2021-12-20:
Hi @wboerhout,

Do you have a link to the Waveshare SX127x you are using? Also, could you please describe the wiring you are using?

wboerhout, 2021-12-20:
waveshare.com/wiki/Pico-LoRa-SX1262-868M

there is no wiring involved, the module is attached to the Pico using pre-soldered pins. see att picture, not much to see I'm afraid.
[image]

wboerhout, 2021-12-20:
and the software came from https://github.com/siuwahzhong/lorawan-library-for-pico as directed by WaveShare

wboerhout, 2021-12-21:
OK. I saw that the PIN layout was wrong, so I scratched the complete software tree and reloaded the ZIP file from https://www.waveshare.com/w/upload/0/08/Pico-LoRa-SX1262-868M_Code.zip

After building, the error was gone. It now loops in "joining the network"  but that is for me to sort out. (network coverage, antenna, etc.)

closed 2021-12-21 by wboerhout