ArmDeveloperEcosystem/lorawan-library-for-pico #14: Raspberry Pi with SX1276 does not join The Things Network

ThomasLeister, 2022-01-06:
## Setup / Environment

- Raspberry Pi Pico (as Single-Wire Debugger, also serving Power via USB) 
- Raspberry Pi Pico (as target)
- SX1276 incl. Breakout board \[3\]
- EU868 location, living in Germany
- Own The Things Network Indoor Gateway (v3) at the location
- Library version: 30d8a0fe7a60bd3456c83b90bb1b7ce5baba41f9

## Goal

Using the SX1276 with the lorawan-library-for-pico on a Raspberry Pi Pico. 
I'd like to use the library's "hello_otaa" example. The device is expected to 
use the OTAA mechanism and the send the "hello world!" message.

I would expect to see the device register at the gateway (gateway logs available
in TTN Gateway Dashboard). After that, I should see a new upstream message in the
Gateway log (and in the device's TTN log).

## Issue

After setting the correct values in config.h, enabling LoRa debugging and uploading the example 
to the Raspberry Pi Pico target, the following log is displayed on the serial console:

```
Pico LoRaWAN - Hello OTAA

Initilizating LoRaWAN ... 
###### =========== CTXS RESTORED =========== ######
Size        : 2208

DevEui      : E6-60-D4-A0-XX-XX-XX-XX
JoinEui     : 00-00-00-00-00-00-00-00
Pin         : 00-00-00-00

success!
Joining LoRaWAN network ... 
###### =========== MLME-Request ============ ######
######               MLME_JOIN               ######
###### ===================================== ######
STATUS      : Duty-cycle restricted
Next Tx in  : 3599891 [ms]
```

_(`XX-XX-XX-XX` was redacted ;-) )_

"Duty-cycle restricted" appears and I cannot observe any new LoRa payload messages or 
"join" messages on the gateway. The device log in TTN Dashboard stays empty. No sign of
communication. TTN Dashboard says "never connected" about the Raspberry Pi Pico.

Even after wainting for an hour (=3599891 ms) nothing happens. I would have expected the 
LoRaWAN stack to re-try the join process, but no more log output appears. Also no updates 
on the TTN dashboard.

## What I've tried

* Triple-checked the wiring, also compared to \[1\] and \[2\]
* Checked Device EUI and App EUI with the TTN gateway.
* Checked frequency range of gateway and SX1276. Both are EU868.
* Checked if my gateway works: It has apparently received some upstream messages by other devices of other LoRa users in the neighborhood. Also tried another public gateway (TTN Indoor Gateway) in the neighborhood
* Tried another SX1276 breakout board / chipset. I have two of those. Same behavior with both. 
* Tried the exact steps at \[2\]. @jbinko seems to have gotten it working. The steps differ a bit from the "official" ones, e.g. he sets `-DACTIVE_REGION=LORAMAC_REGION_EU868`)

I'm pretty clueless why I can't get my setup working. Maybe I forgot something important, 
maybe there were some undocumented changes or I can into a rare bug ;-) 
How could I try to debug the issue / what could possible cause the trouble?

I'm aware that this is not a support forum, but there should be plenty of LoRa experts here,
so that's why I think this might be the most appropriate place to ask for some help. 

## References
- \[1\]: https://www.raspberrypi.com/news/how-to-add-lorawan-to-raspberry-pi-pico/
- \[2\]: https://github.com/jbinko/jbinko.github.io/tree/master/RFM95W
- \[3\]: https://www.ebay.de/itm/254789365056

## Attachments

config.h:

```
/*
 * Copyright (c) 2021 Arm Limited and Contributors. All rights reserved.
 *
 * SPDX-License-Identifier: BSD-3-Clause
 * 
 */

// LoRaWAN region to use, full list of regions can be found at: 
//   http://stackforce.github.io/LoRaMac-doc/LoRaMac-doc-v4.5.1/group___l_o_r_a_m_a_c.html#ga3b9d54f0355b51e85df8b33fd1757eec
#define LORAWAN_REGION          LORAMAC_REGION_EU868

// LoRaWAN Device EUI (64-bit), NULL value will use Default Dev EUI
#define LORAWAN_DEVICE_EUI      "E660D4A0XXXXXXXX"

// LoRaWAN Application / Join EUI (64-bit)
#define LORAWAN_APP_EUI         "0000000000000000"

// LoRaWAN Application Key (128-bit)
#define LORAWAN_APP_KEY         "XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX"

// LoRaWAN Channel Mask, NULL value will use the default channel mask 
// for the region
#define LORAWAN_CHANNEL_MASK    NULL
```

gniezen, 2022-01-07:
I have a very similar setup and I'm experiencing the same issue. I'm using Elementary OS 6.1, so Ubuntu 20.04.

ThomasLeister, 2022-01-12:
Update: I've started using a different LoRaWAN module from Waveshare (SX1262). I'm still in the process of porting their custom version of this library, but I stumbled across the "duty cycle restricted" issue once again.

After reverting Commit e31d3c133f0effca9d3660e35b383506018eca39 and thus removing all new EMMc settings, the library seems to work fine for SX1262.

Maybe Commit e31d3c133f0effca9d3660e35b383506018eca39 is also causing the issue here. ... :thinking:  Need to check that out.

ep1cman, 2022-01-18:
I was unable to replicate this issue with the following setup:

- Adafruit FeatherWing RMF95w (SX1276)
- PicoSDK = 1.3.0
- `LORAWAN_REGION` set to `LORAMAC_REGION_EU868`
- MacOS
- arm-none-eabi-gcc = 10.3.1 20210621
- TTN Indoor Gateway

ThomasLeister, 2022-01-18:
Thanks for trying! Strange. Maybe that issue is related to my particular module type. I guess I'll continue debugging this issue after my WaveShare module has been brought into service :)

But any further feedback / experiences are appreciated! Maybe it helps to hunt down the issue.

ep1cman, 2022-01-18:
Its a long shot but I noticed you used GCC 9 and I used GCC 10. I doubt thats the issue but its worth checking

Mrkvak, 2022-01-20:
I've ran into the same issue today, on a brand new board. After I called `NvmDataMgmtFactoryReset` the problem disappeared. 
More precisely, I've flashed firmware that called the `NvmDataMgmtFactoryReset()` before `lorawan_init_otaa()`. It immediately started to work. Then I flashed the previous version (without the `NvmDataMgmtFactoryReset` call) and it again joined the network successfully.

It looks like something in LoRaMac-node breaks when the NvmDataMgmtRestore fails? In any case, I'm too tired (and probably too unfamiliar with the code) to try to narrow it down today.

ThomasLeister, 2022-01-20:
@Mrkvak this sounds familiar to me. I've run into a very similar issue with my WaveShare SX1262 module. It wouldn't join the network via OTAA and said "Duty-cycle restricted" after a while. I could resolve the issue by running NvmDataMgmtFactoryReset() and then EepromMcuFlush() to write the changes to flash.

To me it seems that there's are bug related to context restoring. Maybe the machanism breaks with invalid data in flash? Not sure what is going on, either, but as soon as I'm back to using my SX1276 module, I'll try and investigate further.

gniezen, 2022-01-21:
I can confirm that doing a `NvmDataMgmtFactoryReset()` fixes the "Duty-cycle restricted" issue.

ep1cman, 2022-01-21:
Do you have to do the reset each time, or after doing it once, does it work from that point forward?

Mrkvak, 2022-01-21:
Like I've said, after performing the NvmDataMgmtFactoryReset once, everything works. Even after subsequent flashes, reboots, etc..

ThomasLeister, 2022-01-21:
Finally had some time to check in this issue. Using `NvmDataMgmtFactoryReset` and `EepromMcuFlush` for a single time resolved the issue for me, too!

I consider this thread solved, then. Still, it might be advisable to mention this pitfall somewhere in the README, because esp. Newbies like me might not be aware that their settings are written to flash and persist on the device, even if the DEV_UID and other "defines" have changed on the meantime.

closed 2022-01-21 by ThomasLeister

sandeepmistry, 2022-02-01:
@Mrkvak, @gniezen, @ThomasLeister thank you for the feedback!

I've created pull request https://github.com/ArmDeveloperEcosystem/lorawan-library-for-pico/pull/18 with a new example that can be used to erase the NVM.