ArmDeveloperEcosystem/lorawan-library-for-pico #19: Problems linking

hardillb, 2022-02-06:
Hi,
I'm trying to build a custom project using this project as a library. I followed the instructions in #11 and while everything compiles I'm having problems at the link stage.

```
...
[build] [ 13%] Linking CXX executable lora-sensor-pico.elf
[build] /usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: CMakeFiles/lora-sensor-pico.dir/src/main.cpp.obj: in function `main':
[build] /opt/share/playing/lora/pico/lora-sensor-pico/src/main.cpp:57: undefined reference to `lorawan_init_otaa(lorawan_sx1276_settings const*, eLoRaMacRegion, lorawan_otaa_settings const*)'
[build] /usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: /opt/share/playing/lora/pico/lora-sensor-pico/src/main.cpp:68: undefined reference to `lorawan_join()'
[build] /usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: /opt/share/playing/lora/pico/lora-sensor-pico/src/main.cpp:70: undefined reference to `lorawan_is_joined()'
[build] /usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: /opt/share/playing/lora/pico/lora-sensor-pico/src/main.cpp:71: undefined reference to `lorawan_process_timeout_ms(unsigned long)'
[build] /usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: /opt/share/playing/lora/pico/lora-sensor-pico/src/main.cpp:90: undefined reference to `lorawan_process_timeout_ms(unsigned long)'
[build] /usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: /opt/share/playing/lora/pico/lora-sensor-pico/src/main.cpp:92: undefined reference to `lorawan_receive(void*, unsigned char, unsigned char*)'
[build] collect2: error: ld returned 1 exit status
[build] make[3]: *** [CMakeFiles/lora-sensor-pico.dir/build.make:1902: lora-sensor-pico.elf] Error 1
[build] make[2]: *** [CMakeFiles/Makefile2:1653: CMakeFiles/lora-sensor-pico.dir/all] Error 2
[build] make[1]: *** [CMakeFiles/Makefile2:1660: CMakeFiles/lora-sensor-pico.dir/rule] Error 2
[build] make: *** [Makefile:186: lora-sensor-pico] Error 2
[build] Build finished with exit code 2
```

I've run `make VERBOSE=1` and the g++ command has the object file for lorawan.c in the list of files to link and running objdump shows the symbols in the file so I'm at a bit of a loss.

All my code is checked in here: https://github.com/hardillb/lora-sensor-pico

Expects pico-sdk and [pimaroni-pico](https://github.com/pimoroni/pimoroni-pico) checked out next to the project

Any suggestions welcome.

sandeepmistry, 2022-02-07:
Hi @hardillb,

You'll need to use:
```c
extern "C" {
#include "pico/lorawan.h"
}
```

in your application for now in your `main.cpp` file for now, I'll look into adding this in `pico/lorawan.h`.

hardillb, 2022-02-07:
Thanks, that now links.

Will have a proper play tonight.

sandeepmistry, 2022-02-07:
Cool, PR https://github.com/ArmDeveloperEcosystem/lorawan-library-for-pico/pull/20 created for this.

sandeepmistry, 2022-03-23:
Closing this as PR #20 has been merged now.

closed 2022-03-23 by sandeepmistry