ArmDeveloperEcosystem/lorawan-library-for-pico #21: 'gpio_set_irq_enabled_with_callback' from incompatible pointer type

DIY62, 2022-02-20:
Hello,

I purchased a Pico-LoRa-SX1262-868M and I'm trying to get it to run with the code-examples from the Waveshare website [https://www.waveshare.com/wiki/Pico-LoRa-SX1262-868M](url) which would be updated to be compatible with the Pico-LoRa-SX1262-868M module
I use the main.c example which is for the sx1262 radio module.

[main.c.txt](https://github.com/ArmDeveloperEcosystem/lorawan-library-for-pico/files/8103914/main.c.txt)

Unfortunately it does not work for me.

While building I got the next warning ...

/home/pi/pico-lorawan/src/boards/rp2040/gpio-board.c: In function 'GpioMcuSetInterrupt':
/home/pi/pico-lorawan/src/boards/rp2040/gpio-board.c:68:73: **warning:** passing argument 4 of 'gpio_set_irq_enabled_with_callback' from incompatible pointer type [**-Wincompatible-pointer-types**]
     gpio_set_irq_enabled_with_callback(obj->pin,GPIO_IRQ_EDGE_RISE,true,**irqHandler**);
                                                                         ^~~~~~~~~~
In file included from /home/pi/pico-lorawan/src/boards/rp2040/gpio-board.c:8:
/home/pi/pico/pico-sdk/src/rp2_common/hardware_gpio/include/hardware/gpio.h:384:103: **note:** expected 'gpio_irq_callback_t' {aka 'void (*)(unsigned int,  long unsigned int)'} but argument is of type 'void (*)(void *)'
 void gpio_set_irq_enabled_with_callback(uint gpio, uint32_t events, bool enabled, **gpio_irq_callback_t callback**);

It seems to be a conflict between the software delivered in the example-code
**(/home/pi/pico-lorawan/src/boards/rp2040/gpio-board.c)**
and pico-sdk
**(/home/pi/pico/pico-sdk/src/rp2_common/hardware_gpio/include/hardware/)**

Despite this issue I copied the uf2-file to the board but it will not connect to the network ...

> Pico LoRaWAN - OTAA - Temperature + LED
> 
> Initilizating LoRaWAN ... success!
> Joining LoRaWAN network ................................

I am not a code-engineer and doing this just for hobby but I would appreciate some help to solve this issue.

Thank you in advance,

Eric

P.S. Is there a example code in micropython or circuitpython which makes it easier to code for people like me ;-)

sandeepmistry, 2022-03-23:
Hi @DIY62,

This library does not currently support the SX1262 that is on the Waveshare board. I suggest you contact the Waveshare team for support.

You can also see @ThomasLeister's fork for the changes needed: https://github.com/ThomasLeister/lorawan-library-for-pico/tree/add-waveshare-sx1262-support

closed 2022-04-06 by sandeepmistry

pinipon1, 2024-05-23:
> Thanks

Hello @DIY62 and everyone seeing this. 
Im having the same issue. Did you solved the problem? Did you change the library?

Thank you in advance,
Diogo

Glenn272, 2024-08-12:
Also having the same issue with the latest Waveshare rp2040 lora example code and it's supposed to support SX1262 now.  Did anyone solve it?
Cheers, Glenn.

peterand, 2025-01-11:
Here is a late suggestion from Perplexity.ai that I have tested and found to work, at least in the ping_pong example:

The warning you're seeing is due to a mismatch in function pointer types. 
The `gpio_set_irq_enabled_with_callback` function expects a callback of type `gpio_irq_callback_t`,
which is defined as `void (*)(unsigned int, long unsigned int)`. 
However, you're passing a function pointer of type `void (*)(void *)`.

To fix this warning, you need to create a wrapper function that matches the expected 
signature and then calls your original callback. Here's how you can modify your code:

1. In your `gpio-board.c` file, add a global variable to store the original callback:

static GpioIrqHandler *g_irqHandler = NULL;

2. Create a wrapper function that matches the expected signature:

static void gpio_irq_wrapper(uint gpio, uint32_t event_mask) {
    if (g_irqHandler != NULL) {
        g_irqHandler(NULL);  // You might want to pass a meaningful context here
    }
}

3. Modify your `GpioMcuSetInterrupt` function:

void GpioMcuSetInterrupt(Gpio_t *obj, IrqModes irqMode, IrqPriorities irqPriority, GpioIrqHandler *irqHandler)
{
    g_irqHandler = irqHandler;
    gpio_set_irq_enabled_with_callback(obj->pin, GPIO_IRQ_EDGE_RISE, true, gpio_irq_wrapper);
}

This solution creates a wrapper function that matches the expected signature and calls your original callback. 
The original callback is stored in a global variable so it can be accessed from the wrapper function.

Keep in mind that this approach assumes you only need one GPIO interrupt at a time. 
If you need to handle multiple GPIO interrupts simultaneously, 
you'll need to implement a more sophisticated solution, such as maintaining a map of GPIO pins to their 
respective callbacks.