ArmDeveloperEcosystem/lorawan-library-for-pico #23: How to configure for use with SX1262 Semtech Shield

NicoPowers, 2022-04-04:
Hello,

I am trying to use this package for the SX1261/1262 with the Pico, but I am having a hard time trying to figure out how to get it to compile safely for the SX1262 instead of the SX1276; is this documented anywhere? If not, would this be possible to implement?

Thank you

sandeepmistry, 2022-04-04:
Hi @NicoPowers,

This library currently does not support the SX1276, however pull request #15 is open to add support for it.

BNNorman, 2022-05-22:
waveshare have a variant which supports sx1262 try here:-

https://www.waveshare.com/wiki/Pico-LoRa-SX1262

BUT, I have not gotten it to use my keys and have opened a support ticket. The hardware transmits ok - I can see it in my gateway but it isn't using the keys I put in the config.h for the otaa_temperature_led example.

BNNorman, 2022-07-21:
Remiss of me not to post that I fixed it. The problem was with the compilation of "sscanf" which wasn't scanning the keys properly (at all). I replace the sscanf with my own code and that fixed it. I reported the issue to Waveshare but don't know if they made changes to their code. In fact the demo code link on their wiki appears broken, If anyone want to know my fix let me know and I'll post it.

BNNorman, 2022-07-21:
I posted my working code here.
There were two problems: The original code didn't allow spi1 to be used. I fixed that and Sandeep Mistry has now amended his code.
Second the pico compilation of sscanf didn't work as expected - I checked that with a really simple program and it failed so I wrote a replacement (simple) key scanner. It's all documented in the following

https://github.com/BNNorman/Waveshare-Pico-LoRa-SX1262-868M

closed 2022-09-26 by stale[bot]