ArmDeveloperEcosystem/lorawan-library-for-pico #24: spi1 does not work

BNNorman, 2022-05-22:
I'm trying to use an ILABS CHALLENGER RP2040 Lora 868M module. This has an RFM95 on-board with SX1276 chip and uses spi1 not spi0.

I'm tryinmg to get the otaa_temperature_led example working on it

in main.c I have this.
```
// pin configuration for SX12xx radio module
const struct lorawan_sx12xx_settings sx12xx_settings = {
    .spi = {
        .inst = spi1,
        .mosi = 11,
        .miso = 12,
        .sck  = 10,
        .nss  = 3
    },
    .reset = 15,
    .busy = 2,
    // sx127x would use dio0 pin, and sx126x dont use it 
    // .dio0  = 7,
    .dio1  = 20
};
```
This works fine using a simple SPI program to read the sx1276 Version (0x12).

However when using pico-lorawan your code blocks and never returns a value. The reason is in spi-board.c  here:- 

```
uint16_t SpiInOut( Spi_t *obj, uint16_t outData )
{
    const uint8_t outDataB = (outData & 0xff);
    uint8_t inDataB = 0x00;

    spi_write_read_blocking((obj->SpiId == 0) ? spi0 : spi1, &outDataB, &inDataB, 1);

    return inDataB;
}
```
obj->SpiId returned zero instead of 1 when I added print statements. It seems that the 'obj' hasn't been updated somewhere along the line. (I have to break off now but will hunt for that later)

Regards
Brian

BNNorman, 2022-05-22:
Adding this at the start of SpiInit() in src/boards/rp2040/spi-board.c works
```
obj->SpiId=spiId;
```

tnakasaki, 2022-05-24:
Hi @BNNorman, I'm happy to see someone using this board as I have just purchased one and could not get it to work with this repository's examples.
Where did you get the pin assignments? I am referencing https://ilabs.se/challenger-rp2040-lora-datasheet/. The nss, reset, and dio1 pins on the data sheet differ from the pins specified in your code.

BNNorman, 2022-05-24:
The pins are mentioned in that document if you scroll past the pinout diagram. The lora pins are not exposed. I also found the schematic which confirmed those gpio pin assignments. It was the spi1 implementation in the software which was causing the problem on this board.

tnakasaki, 2022-05-24:
@BNNorman The data sheet lists

> GPIO14, GPIO15 and GPIO18 is connected to DIO0 – DIO2 respectively of the RFM95W.

but your code says

> .dio1  = 20

Pin 20 is broken out for use on the board, physically labeled "SDI".

Also, the datasheet says 

> "GPIO9 – LORA_CS"

while your code says

> .nss  = 3

Pin 3 is also broken out on the board.

The sheet says
> GPIO13 is connected to the reset input of the RFM95W.

Your code lists

> .reset = 15

I thought GPIO15 should be the reset.

This is why I'm confused

BNNorman, 2022-05-24:
> <ArmDeveloperEcosystem/lorawan-library-for-pico/issues/24/1135489201@
> github.com>

Hmm, ok. I'll double check when I get back home. Probably better to PM me
as this goes beyond the library issue I raised.

tnakasaki, 2022-05-24:
@BNNorman Thank you for clarifying. I'm sorry, I did not intend to sound rude. I just wanted us to have the same starting point.
Cheers

sandeepmistry, 2022-05-30:
@BNNorman would you be able to open a pull request for the change you suggested in https://github.com/ArmDeveloperEcosystem/lorawan-library-for-pico/issues/24#issuecomment-1133907859 ?

BNNorman, 2022-05-30:
I have forked your repository with the aim of trying to get it to work with the sx126x radio since Waveshare's  pico-lorawan 
 code changes ignore my TTN keys even though it is based on your repo. (Which works fine on other HDW using sx1272 thatI have)

I also changed the default REGION to EU868 in the OTAA_temperature_LED example config.h. 

If I create a pull request I think it will include both changes. 

As you can see from the screen shot, there is just one addition after line 14. Is it worth the effort of a pull request just for that? 

[image]

sandeepmistry, 2022-05-30:
> As you can see from the screen shot, there is just one addition after line 14. Is it worth the effort of a pull request just for that?

It's up to you, I was just thinking it would be nice for you to get credit for the fix.

BNNorman, 2022-05-30:
I'm not that bothered. But thanks for the thought.

PR #26 (2022-06-09): Assign SPI id in SpiInit
Resolves #24.

As suggested by @BNNorman in https://github.com/ArmDeveloperEcosystem/lorawan-library-for-pico/issues/24#issuecomment-1141217289.
closed 2022-06-09 by sandeepmistry