ArmDeveloperEcosystem/lorawan-library-for-pico #32: Complier error

machasm1, 2023-02-28:
Getting a complier error here on line 295 of lorawan.c file
`int lorawan_process_timeout_ms(uint32_t timeout_ms)`

Error message is this
a value of type "int" cannot be used to initialize an entity of type "absolute_time_t"
Any ideas?

jerryneedell, 2023-02-28:
In lorawan.c try adding "#include pico/time.h"

```
diff --git a/src/lorawan.c b/src/lorawan.c
index 9e4cec3..dc24e60 100644
--- a/src/lorawan.c
+++ b/src/lorawan.c
@@ -28,6 +28,7 @@
 #include <string.h>
 
 #include "pico/lorawan.h"
+#include "pico/time.h"
 
 #include "board.h"
 #include "rtc-board.h"
```

closed 2023-02-28 by machasm1

jerryneedell, 2023-02-28:
Glad it worked -- I created PR #33 to make this fix available to others.