ArmDeveloperEcosystem/lorawan-library-for-pico #35: Support for adafruit_feather_rp2040_rfm9x

mdede439, 2023-05-12:
Adafruit recently developed and released a new RP2040 board with a LoRa module installed.  It is different than the standard RP204 with an externally attached LoRa module.  Is there an easy way to modify the definition of  "-DPICO_BOARD=adafruit_feather_rp2040" to the newer board? 

When I use the otaa_temperature_led example it fails to initialize the the lorawan.  
I get this error; 

Pico LoRaWAN - OTAA - Temperature + LED

Initilizating LoRaWAN ... failed!!!

I am pretty sure it is because of the design differences in the two boards.

mdede439, 2023-05-13:
I was able to get the new Adafruit RP2040_RFM95 board to work by modifying just the GPIO pins used for the LoRa module.  I thought there would be more to it!

const struct lorawan_sx1276_settings sx1276_settings = {
    .spi = {
//        .inst = PICO_DEFAULT_SPI_INSTANCE,
        .mosi = 15,
        .miso = 8,
        .sck  = 14,
        .nss  = 16
    },
    .reset = 17,
    .dio0  = 21,
    .dio1  = 22
};

sandeepmistry, 2023-05-15:
@mdede439 great work!

Have you asked Adafruit if they plan to add a .h here for the board in the `pico-sdk`? https://github.com/raspberrypi/pico-sdk/tree/develop/src/boards/include/boards

closed 2023-07-21 by stale[bot]