ArmDeveloperEcosystem/lorawan-library-for-pico #41: second join always ends in rx 2 timout

smalltoe, 2023-08-31:
I have this reproducable pattern:
Step 1: I reset pico with flash_nuke.uf2
Step 2: I flash and run otaa_temperature_led.uf2
Step 3: I disconnect/reconnect power from pico and run otaa_temperature_led.uf2

Step 1 and Step 2 run flawless (DevEui is masked):
```
Pico LoRaWAN - OTAA - Temperature + LED

Erasing NVM ... success!
Initilizating LoRaWAN ... DevEui      : XX-XX-XX-XX-XX-XX-XX-XX
JoinEui     : 00-00-00-00-00-00-00-00
Pin         : 00-00-00-00

success!
Joining LoRaWAN network ...
###### =========== MLME-Request ============ ######
######               MLME_JOIN               ######
###### ===================================== ######
STATUS      : OK
........
###### =========== MLME-Confirm ============ ######
STATUS      : OK
###### ===========   JOINED     ============ ######

OTAA

DevAddr     :  01481CFE

DATA RATE   : DR_0

###### ============ CTXS STORED ============ ######
Size        : 2208

. joined successfully!
sending internal temperature: 20 °C (0x14)...
###### =========== MCPS-Request ============ ######
######           MCPS_UNCONFIRMED            ######
###### ===================================== ######
STATUS      : OK
success!

###### =========== MCPS-Confirm ============ ######
STATUS      : OK

###### =====   UPLINK FRAME        1   ===== ######

CLASS       : A

TX PORT     : 2
TX DATA     : UNCONFIRMED
14

DATA RATE   : DR_0
U/L FREQ    : 867300000
TX POWER    : 0
CHANNEL MASK: 00FF

```
In step 3 the device cannot join:
```
Pico LoRaWAN - OTAA - Temperature + LED

Erasing NVM ... success!
Initilizating LoRaWAN ... DevEui      : XX-XX-XX-XX-XX-XX-XX-XX
JoinEui     : 00-00-00-00-00-00-00-00
Pin         : 00-00-00-00

success!
Joining LoRaWAN network ...
###### =========== MLME-Request ============ ######
######               MLME_JOIN               ######
###### ===================================== ######
STATUS      : OK
.......
###### =========== MLME-Confirm ============ ######
STATUS      : Rx 2 timeout
############################################# cut out a couple of rows here as it is repeated 12 times.

###### =========== MLME-Request ============ ######
######               MLME_JOIN               ######
###### ===================================== ######
STATUS      : Duty-cycle restricted
Next Tx in  : 3507312 [ms]

###### ============ CTXS STORED ============ ######
Size        : 2208

......
```
The only point that helps at this point is restart at Step 1 from above. - or wait for an hour.
Are those symptoms related to the gateway or the node? I am using RAK7289CV2.
If a pro could give me a hint where to look next or some context I would be very grateful.

smalltoe, 2023-08-31:
Thinking through the topic I guess I have found the right approach and understanding:
Joining the network has to be executed only once, like in step 2. After that the node has joined and does not need to join again.
With `lorawan_is_joined()` I can check if the node has joined.
With a slight modification of the example like the below, I can check if the node joined already and only if that is not the case execute the procedures to join.
```
    if (!lorawan_is_joined()) {

        // Start the join process and wait
        printf("Joining LoRaWAN network ...");
        lorawan_join();

        while (!lorawan_is_joined()) {
            lorawan_process_timeout_ms(1000);
            printf(".");
        }
        printf(" joined successfully!\n");
    }
```

smalltoe, 2023-09-15:
Looks like that was not it. After batteries drained the behaviour returned.

ioanfesteu, 2023-11-21:
Your problem seems to have to do with the duty cycle. I see you are using EU868 where duty cycle is restricted to 1% of total airtime. Try edit the folowing line in "src/lorawan.c" file by changing true to false then try again. :
```
`/*!
 * LoRaWAN ETSI duty cycle control enable/disable
 *
 * \remark Please note that ETSI mandates duty cycled transmissions. Use only for test purposes
 */
#define LORAWAN_DUTYCYCLE_ON                        true`
```

I encountered the same problem while testing my node and I wasn't patient enaugh.

Smartich0ke, 2024-06-01:
I'm also having this same problem. Setting `LORAWAN_DUTYCYCLE_ON` to false didn't help and neither did the join logic by @smalltoe . In the TTN console I get the occasional MIC Mismatch error. I have double checked and triple checked the app-eui, dev-eui, and join-eui to be correct.

Smartich0ke, 2024-06-29:
I'm still having the problem and haven't found a solution. Could anyone help with this?

s00500, 2024-07-03:
Seems like I have exactly the same issue here...

PatersonJQ, 2025-06-07:
Just thinking out loud here feel free to ignore. @smalltoe, what’s your gateway say? In the events you should see an error message.

Have you ensured your device NONCE is not that same?