ArmDeveloperEcosystem/lorawan-library-for-pico #42: frame counter lost after power cycle?

ioanfesteu, 2023-11-21:
Hi, I would like to ask for help for my problem:

I am running the library without problems but  I am at the point I am looking to save battery energy. I managed to put my Pico in deep sleep mode but even in such a state it consumes about 1.8 mA which is way to much. So I was thinking to use an auxiliary mcu such as ATTiny85 to power off my Pico between  sensor readings (TinyML stuff) and LoRaWAN sessions then going in deep sleep mode (around 0.0047 mA).

After a power cycle I am not going to join again the network but instead I am invoking only `lorawan_init_otaa()` then send my payload when ready. Can anybody confirm that the` lorawan_init_otaa() `or `lorawan_init()` are reseting the `FCntUp` to 0 or they are retrieving the last saved value (if saved)?

I have some succes with the above mentioned method but some uplinks are skipped until apparently the nodes `FCntUp` is catching up with the network server one. Of course I can Join the network every time after a power cycle but is not the preffered solution nor elegant one.

Thank you!

ioanfesteu, 2023-11-28:
I can confirm that `FCntUp ` is stored. It was my code that messed things.
So, if anybody wants to power off the Pico to conserve battery this can be done with an external mcu then at startup only invoke `lorawan_init_otaa()` , whitout the join steps, and the context will be restored from the nvram.

sandeepmistry, 2023-12-04:
@ioanfesteu thanks for the update, can this be closed now?

ioanfesteu, 2023-12-04:
@sandeepmistry yes it can be closed, thank you.

closed 2023-12-04 by sandeepmistry